timlaing/modbus_local_gateway #12: No updates with update frequency below 30s

Pel1can111, 2023-12-13:
Everything seems to work fine when using the default update frequency of 30s but reducing this to 10s seems to stop values updating. At first everything works fine but after a few minutes values stop updating.

Using https://www.waveshare.com/product/rs232-485-422-to-poe-eth-b.htm to connect to a Eastron SDM230

Curious if this is just a hardware limitation or something else.

timlaing, 2024-01-01:
I run this with a 1 sec update interval with no issues to an Easton SDM230 using https://www.waveshare.com/product/iot-communication/wired-comm-converter/ethernet-to-uart-rs232-rs485/rs485-to-wifi-eth.htm.

Are you able to increase the logging on pymodbus & modbus_local_gateway to help diagnose this issue.

Thanks.

Lembi203, 2024-01-17:
I have the same problem, it works for the first minute then it stops updating, it then works only sporadically.
the strange thing is, when i go to the configuration webpage or reload it, it is updating the values again for a short time

timlaing, 2024-01-17:
are you able to provide any more details on your configuration or debug output in order to fault find.

Lembi203, 2024-01-17:
[image]

I get this errors 
And i use the Waveshare rs485 to eth (b), but should be basically the same just without wifi.

Lembi203, 2024-01-17:
[home-assistant_modbus_local_gateway_2024-01-17T23-30-00.228Z.log](https://github.com/timlaing/modbus_local_gateway/files/13969856/home-assistant_modbus_local_gateway_2024-01-17T23-30-00.228Z.log)

This is the Debug Log, it is only a fiew seconds, at first it was working fine, then it stopped working.

Do not wonder, i removed some entities from the yaml file, i do not need so much informations from the power meter, but it is the same case with the original yaml file

timlaing, 2024-08-08:
This should be resolved in release v1.4.0

closed 2024-08-08 by timlaing