timlaing/modbus_local_gateway #19: Closing connection - Device not available

sftgunner, 2024-04-27:
Currently getting the error `[custom_components.modbus_local_gateway.tcp_client] Closing connection - Device not available AsyncModbusTcpClientGateway 10.0.22.250:502 [1]`.
I am using the same WaveShare Wi-Fi to RS485 as recommended, along with my Heatmiser Edge thermostat. I've got a very barebones yaml set up, and keep getting the error above. I initially saw it working briefly, but haven't seen it function again. The native HomeAssistant MODBUS integration continues to be able to interact with the device fine. Any idea what might be causing this?

Basic yaml:
```
device:
  manufacturer: Heatmiser
  model: Edge
  operation_mode:
    title: Current operation mode
    address: 33
    control: number
    icon: mdi:serial-port
    precision: 0
    number:
      min: 0
      max: 5

entities:
  machine_status:
    title: Code version number
    address: 1
```

AravindPrabhs, 2024-05-09:
How are you configuring the gateway module ? I wasn't able to get the device to connect (over the standard 502 port). My configuration step was setting the data transfer mode to "modbus tcp <=> modbus rtu" and then configuring the uart settings appropriately, then restarting. all using the web gui.

sftgunner, 2024-08-08:
Hi Tim,
Just seen the bump to 1.4.0 (hopefully it will resolve any issues relating to this original ticket!), but I'm getting this error after updating:
Config flow could not be loaded: {"message":"Invalid handler specified"}

Is this expected/known issue, or something to do with my specific setup?

I've just updated to 2024.8 as well, not sure if that makes a difference. I've also tried removing from HACS, rebooting and then reinstalling via HACS, to no avail...
Can open a new issue to track this specific problem, if preferred.

timlaing, 2024-08-08:
I'll take a look at this for you.

sftgunner, 2024-08-08:
Fab, thanks v much! If it's at all helpful, I'm happy to debug/collect logs as required to aid with troubleshooting

sftgunner, 2024-08-08:
Just realised I forgot to do the most basic troubleshooting step - here is the error in HA logs:

```Logger: homeassistant.config_entries
Source: config_entries.py:2871
First occurred: 15:32:41 (1 occurrences)
Last logged: 15:32:41

Error occurred loading flow for integration modbus_local_gateway: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.12/site-packages/pymodbus/framer/__init__.py)
```

timlaing, 2024-08-08:
@sftgunner - I will be creating a hot fix release shortly which will address this issue.

sftgunner, 2024-08-08:
Many thanks for your rapid changes! I have updated to 1.4.1, but unfortunately the issue persists, with the same error:
```
Logger: homeassistant.config_entries
Source: config_entries.py:2871
First occurred: 20:35:11 (1 occurrences)
Last logged: 20:35:11

Error occurred loading flow for integration modbus_local_gateway: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.12/site-packages/pymodbus/framer/__init__.py)
```
I've checked with VS code to ensure that the HACS update has gone through successfully (i.e. that the code changes match the raw files in custom_components), and made sure I did a reboot after this, but I'm not sure if there's a way from HASSOS to check that pymodbus is definitely installed at version 3.7.

timlaing, 2024-08-08:
I'll take another look at this for you. Sorry this is causing you problems

timlaing, 2024-08-08:
@sftgunner - I am struggling to recreate your issue. Can you try updating the `/config/custom_components/modbus_local_gateway/manifest.json` and changing the following line: `"pymodbus~=3.7",` to `"pymodbus==3.7.0",`
If that works for you I'll create a new hotfix. 

Thanks

sftgunner, 2024-08-09:
No worries, thanks very much for your help!
I've made the suggested change, but unfortunately I still appear to be getting the same error. I'll keep digging; Google is not throwing up any results as to how I can verify that pymodbus is definitely installed at the correct version, which is very tedious.

timlaing, 2024-08-09:
I think i have found the issue. If you are using the core modbus component, this is still requiring version 3.6.9 as such is preventing the upgrade for this component. I will downgrade to improve compatibility with this component.

timlaing, 2024-08-09:
I have published v1.4.2 which should resolve your issues.

sftgunner, 2024-08-09:
Hi Tim, thanks very much indeed! Can confirm this has resolved the issue with not being able to set up device. Unfortunately, the original issue 
> Closing connection - Device not available AsyncModbusTcpClientGateway 10.0.22.250:502 [1] 

persists, but I suspect this might be to do with the Waveshare setup.
Is there a recommended firmware version for this? Mine is still on 1.452, but I've seen discussion (https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/17) that 1.486 might be more reliable in some circumstances.

Either way, I will verify my setup matches the README shortly, and will re-open this issue if it still seems to be a problem after that.

closed 2024-08-09 by sftgunner