timlaing/modbus_local_gateway #38: Eastron SDM630 v2 with Waveshare rs485

SpeijerTom, 2025-03-24:
Hello, 

I'm using your integration to get data from my sdm, which for 90% works great.
So thank you for that.
I have an issue with some values being unavailable, but not always.

Particularly the following : 
They happen to be the values I use for my Energy Dashboard

- sensor.eastron_total_export_active_energy 
- sensor.eastron_total_import_active_energy 

[image]

any idea what might be the issue?
Is there anything I can do to give you some more information?

I've also added the latest logs.

[home-assistant_modbus_local_gateway_2025-03-24T08-51-19.728Z.log](https://github.com/user-attachments/files/19426810/home-assistant_modbus_local_gateway_2025-03-24T08-51-19.728Z.log)

timlaing, 2025-03-25:
I think I have found the issue here and will be releasing an update shortly.

timlaing, 2025-03-26:
Based on my testing I believe this issue has now been resolved. If you wish to test this before I create the release, update to the version on the main branch.

SpeijerTom, 2025-03-26:
Hey Tim, thank you for your fast action :-)
As I'm not sure how to update through the main branch, I'll wait for the new release.

Unless you want me to test it, please explain how I can update it with the version on the main branch.

Thank you

timlaing, 2025-03-26:
I have created the new release. Can you check that this resolves your issue. Thanks.

SpeijerTom, 2025-03-26:
Hey Tim, just installed the new version and currently there are no unavailable values, so, so far so good.
Will keep you posted, thank you again for all the contributions

timlaing, 2025-03-27:
Closing this issue as resolved.

SpeijerTom, 2025-03-31:
Tim, Sorry to re-open this.

The "unavailable" issue seems to be resolved, haven't had an unavailable since the update.
But, after looking at the numbers, and comparing it with the values seen on the display.
The IMPORT/EXPORT (total_export_active_energy, total_import_active_energy) values are way off..
example, when i see 1600 Kwh in Home assistant, on the display it shows 1,6 Kwh
At first glance, it looks like it needs to be divided by 100, but when looking at the history off the values, the export for example is showing that during the night, it's also exporting... obviously that's not possible, except perhaps with moonlight :-) 

Here 2 examples : 
This shows an export of 1481 kwh between 2 and 3 am in the night
[image]

This shows an import of 7384 kwh between 3 and 4 am in the night
[image]

any idea?
All the rest of the values are spot on when comparing them with the values on the display

timlaing, 2025-03-31:
@SpeijerTom  I'll take a look for you.

timlaing, 2025-03-31:
@SpeijerTom - if you could try updating to the main branch and giving this a test that would be great.

In Home Assistant 
 > Developer tools > Actions

Action: Update: Install update
Targets: Integration for Modbus TCP Gateways update (Entity)
Version: main

> Perform action

Hope this helps.

https://www.home-assistant.io/docs/tools/dev-tools/

timlaing, 2025-04-01:
Thanks - this is useful, as I don't have a SDM630 to test with only a SDM230. I will have a better look this evening.

SpeijerTom, 2025-04-01:
You are welcome, happy to help you improve the integration because it will help a lot of people.
Let me know when I need to test something else.

SpeijerTom, 2025-04-07:
Hey Tim, did you manage to find some time to take a look at it?
Thank you

SpeijerTom, 2025-04-09:
Tim, one issue has already been resolved... when updating the available version for the modbus_local_gateway to the latest, it now works. I didn't do that, because i thought it was the same version i used from the main branch.. 

so this is ok now : 

[image]

SpeijerTom, 2025-04-21:
Hey Tim, did you manage to take a look at it? Thank you

SpeijerTom, 2025-04-21:
Tim, Sorry, didn't knew that. Family and Holiday are 2 sacred things. Take your time 😉

timlaing, 2025-05-02:
@SpeijerTom - I have managed to recreate part of you issue (random peaks overnight). Still trying to find the cause, hopefully will have a fix for this soon.

SpeijerTom, 2025-05-06:
Hey Tim, thank you for adding the max change, I updated my entire HA just now, including your update.
However I now (even after a couple of restarts) get this : 

1 ERROR

`Logger: homeassistant.setup
Source: setup.py:340
First occurred: 2:08:31 PM (1 occurrences)
Last logged: 2:08:31 PM

Setup failed for custom integration 'modbus_local_gateway': Unable to import component: cannot import name 'TransactionManager' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 992, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/modbus_local_gateway/__init__.py", line 20, in <module>
    from .coordinator import ModbusCoordinator
  File "/config/custom_components/modbus_local_gateway/coordinator.py", line 22, in <module>
    from .tcp_client import AsyncModbusTcpClientGateway
  File "/config/custom_components/modbus_local_gateway/tcp_client.py", line 15, in <module>
    from .transaction import MyTransactionManager
  File "/config/custom_components/modbus_local_gateway/transaction.py", line 7, in <module>
    from pymodbus.transaction import TransactionManager
ImportError: cannot import name 'TransactionManager' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1012, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1004, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/modbus_local_gateway/__init__.py", line 20, in <module>
    from .coordinator import ModbusCoordinator
  File "/config/custom_components/modbus_local_gateway/coordinator.py", line 22, in <module>
    from .tcp_client import AsyncModbusTcpClientGateway
  File "/config/custom_components/modbus_local_gateway/tcp_client.py", line 15, in <module>
    from .transaction import MyTransactionManager
  File "/config/custom_components/modbus_local_gateway/transaction.py", line 7, in <module>
    from pymodbus.transaction import TransactionManager
ImportError: cannot import name 'TransactionManager' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)`

and 1 WARNING

`Logger: homeassistant.util.loop
Source: util/loop.py:77
First occurred: 2:08:31 PM (1 occurrences)
Last logged: 2:08:31 PM

Detected blocking call to import_module with args ('custom_components.modbus_local_gateway',) in /usr/src/homeassistant/homeassistant/loader.py, line 1052: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, pl [... cut ...] ", line 171, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1004, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)`

timlaing, 2025-05-06:
> Fixed it for now...
> When doing al the updates, I also updated the HA Core version to 2025.4.4 
> I now reverted back to 2025.4.3
> 
> and it works again...

Not sure why 2025.4.4 is broken but I'll get a fix out for that soon.

SpeijerTom, 2025-05-07:
Tim, thanks for developing the fix for 2025.4.4 

Do you have a discord or something?
By using that, you can close this issue and I keep you informed of things that break?

BTW : Did you change something on the Export Energy? when applying the MAX fix?
Because that value has become "unavailable" again 

[image]

[image]

Thx

SpeijerTom, 2025-05-19:
Hey Tim, 

You are welcome. (discord => tomspeijer_50068 )
I installed the update, but the export value stays "unavailble", it is the only value which is unavailable btw..

SpeijerTom, 2025-05-20:
strange : 

[image]

SpeijerTom, 2025-05-27:
Hey Tim, 

Installed the new update and enable debug logging.
The values are unavailable due to max values exceed the 500 delta?
Trying to figure out why sometimes one of the 2 used to work and then all of a sudden they get (like my previous comment) negative values..
[image]

btw i've sent you a request in discord

dmatscheko, 2025-06-08:
Hi @SpeijerTom, I had a similar issue a while ago and it went away on its own after I updated Home Assistant. Can you try if the value is correct, if you run this test script? If the output is not correct, could you post the output of the script?

[https://github.com/dmatscheko/modbus_connect/blob/main/support/modbus_test_client/test.py](https://github.com/dmatscheko/modbus_connect/blob/main/support/modbus_test_client/test.py)
(Only this script, you don't need to check out the whole integration.)

If you want to run the test script, please change MODBUS_HOST, MODBUS_PORT, and SLAVE_ID to your device and REGISTER_ADDRESS to the register that has too high values.
Also, please do NOT set ENABLE_WRITE to true.

Hope that helps to find out the reason for this behavior.

In my case, the test script reported the wrong values too, but now that everything works, I can't get it to report the wrong numbers again.

closed 2025-06-08 by timlaing