timlaing/modbus_local_gateway #52: Device with everything in Holding Registers

maniac-semal, 2025-05-15:
Hello,

Question:
As far as I can tell from the list I have attached, the Heat Pump that I am using is placing everything in holding registers. This also includes Binary sensors. 

The only way I have got it to work in general is by assigning everything to "read_write_word", otherwise I always get an "count does not match" error. I have no general issue in reading or also writing these values as long as I stay with the read_write_word.

The only problem is that I cannot assign the binary sensors or switches to actual binary sensors or switches in home assistant. Since you seem to Automatically assign read_write_words to a "sensor". (or select/switch/number for commands).
Could you add an option to set a read_write_word to binary_sensor or switch?

Here is a short extract of a working config (only just started gathering all the values I want to use):

```

device:
  manufacturer: Blaupunkt
  model: BLP15P3V2MR32

read_write_word:

  ###################
  ##### Sensors #####
  ###################

  inlet_temperature:
    name: "Rücklauf Tmperatur"
    address: 0x000E
    precision: 1
    multiplier: 0.1
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:water-thermometer

  outlet_temperature:
    name: "Vorlauf Tmperatur"
    address: 0x0012
    precision: 1
    multiplier: 0.1
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:water-thermometer

  ambi_temperature:
    name: "Umgebungstemperatur Tmperatur"
    address: 0x0011
    precision: 1
    multiplier: 0.5
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:sun-thermometer

  ###################
  ##### Configs #####
  ###################

  heating_hysteresis:
    name: Heizen Hysterese
    address: 0x00C6
    precision: 1
    multiplier: 1
    control: number
    number:
      min: 2.0
      max: 18.0
    unit_of_measurement: °C
    icon: mdi:home-thermometer-outline
    entity_category: config
    entity_registry_enabled_default: True

  unit_mode:
    name: "Betriebsmodus"
    address: 0x0036
    control: select
    options:
      0: Warmwasser
      1: Heizen
      2: Kühlen
      3: Heizen und Warmwasser
      4: Kühlen und Warmwasser
    icon: mdi:cog
    entity_category: config
    entity_registry_enabled_default: True

  water_hysteresis:
    name: Warmwasser Hysterese
    address: 0x00C8
    precision: 1
    multiplier: 1
    control: number
    number:
      min: 2.0
      max: 18.0
    unit_of_measurement: °C
    icon: mdi:home-thermometer-outline
    entity_category: config
    entity_registry_enabled_default: True

  pump_work:
    name: "Betriebsmodus Pumpe"
    address: 0x019E
    control: select
    options:
      0: Intervall
      1: Normal
      2: Bedarf
    icon: mdi:pump
    entity_category: config
    entity_registry_enabled_default: True

  dc_pump_mode:
    name: "Modus Pumpe"
    address: 0x019E
    control: select
    options:
      0: Manuell
      1: Auto
    icon: mdi:pump
    entity_category: config
    entity_registry_enabled_default: True

  compressor_mode:
    name: "Kompressor Modus"
    address: 0x0190
    control: select
    options:
      0: Normal
      1: ECO
      2: Nacht
    icon: mdi:car-brake-low-pressure
    entity_category: config
    entity_registry_enabled_default: True
  ###################
  ### Diagnostics ###
  ###################

  suct_gas_temp_temperature:
    name: "Gas Einlass Tmperatur"
    address: 0x0015
    precision: 1
    multiplier: 0.5
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:thermometer-lines
    entity_category: diagnostic
    entity_registry_enabled_default: True

  coil_temperature:
    name: "Verdampfer Temperatur"
    address: 0x0016
    precision: 1
    multiplier: 0.5
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:thermometer-lines
    entity_category: diagnostic
    entity_registry_enabled_default: True

  exhaust_temperature:
    name: "Abgas Temperatur"
    address: 0x001B
    precision: 1
    multiplier: 0.5
    device_class: temperature
    unit_of_measurement: °C
    icon: mdi:thermometer-lines
    entity_category: diagnostic
    entity_registry_enabled_default: True

read_only_word:

  hotwater_demand:
    name: "Heißwasser Nachfrage"
    address: 0x0003
    bits: 1
    device_class: heat
    icon: mdi:heating-coil
    entity_category: diagnostic
    entity_registry_enabled_default: True

  heating_demand:
    name: "Heizung Nachfrage"
    address: 0x0003
    bits: 1
    shift_bits: 1
    device_class: heat
    icon: mdi:heating-coil
    entity_category: diagnostic
    entity_registry_enabled_default: True

read_write_boolean:

read_only_boolean:

```

[MODBUS_PROTOCOL_CG248066--CG248067 (2).pdf](https://github.com/user-attachments/files/20234602/MODBUS_PROTOCOL_CG248066--CG248067.2.pdf)

maniac-semal, 2025-05-16:
Hello Again,
I have been playing Around with this today and managed to partially get some things working the way I want. While doing that I noticed something in your implementation of the "bit flags" which I can use for some of my holding registers (spefically the Failure registers).
Generally that works, for example like this:

```
  failure_symbol_1:
    name: Alarme 1
    address: 0x0007
    flags:
      1: Hotwater temp
      2: Hotwater temp
      4: coil temp.
      5: Outlet temp.
      6: High pressure sensor failure
      8: High pressure sensor failure
    bits: 8
    icon: mdi:message-alert-outline
    entity_category: diagnostic
```

Only issue is that when there is no active bit in there, then Home Assistant will deem the "Sensor" as "Not available". As soon as one of the flags is active. It works as designed...

The issue seems to be that the read value in log is "None":

> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key output_symbol_1 is Lüfter Geschwindikeit Hoch
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key output_symbol_2 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key output_symbol_3 is A/C PUMP
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_1 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_2 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_3 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_4 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_5 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_6 is None
> 2025-05-16 23:28:29.308 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key failure_symbol_7 is None

These here are all sensors with flags in my config. The ones with "None" are displayed as "Not Available in Home Assistant. So it seems somewhere from here to Home Assistant something doesnt like the "None".

I played around a little and managed to it to work "better" (Depending who you ask) by modifying the conversion.py in the function "_convert_to_flags". Maybe you have a more elegant solution (which will also be translated...), be my guest 😉 

```
    def _convert_to_flags(
        self, registers: list, desc: ModbusEntityDescription
    ) -> str | None:
        """Convert to a flags type"""
        int_val: int = int(self._convert_to_decimal(registers=registers, desc=desc))
        ret_val: str | None = None
        if desc.conv_flags:
            for key in desc.conv_flags:
                if int_val & (0x1 << (key - 1)) != 0:
                    if ret_val is None:
                        ret_val = desc.conv_flags[key]
                    else:
                        ret_val += f" | {desc.conv_flags[key]}"
            if ret_val is None:
                ret_val = "No Messages"      <--------- Sadly not Translated and maybe not the best way...But it seems to work
            return ret_val
```

timlaing, 2025-05-18:
@maniac-semal - thanks for your input. I will take a look at this for you.

timlaing, 2025-05-18:
In order to map a holding register to a switch in home assistant use the following yaml:
```
  on_off:
    name: On/Off
    address: 0
    control: switch
    icon: mdi:switch
    switch:
      "off": 0
      "on": 1
```

For you flags issue - you need to map in a value for the normal status, for example:
```
  status:
    name: Status
    address: 2
    map:
      0: Standby
      1: Normal
      3: Fault
      4: Flash
    bits: 8
```

Hope this helps.

maniac-semal, 2025-05-18:
Hello timlaing,

sadly. Both do not really work. First of the switch. Problem in my case is that the switch is only one part of a holding register, so one bit.

Here is an Extract of the PDF I attached in the beginning:

[image]

My "Switch" is for example bit 0. If I map it to the single bit 0 of address 0x0032 then Home-Assistant reads it, but as soon as I try to activate it, Home Assistant says "cannot write single bit" or something like that.
If I adress it to the 0x0320 that is mentioned its also not working.

If I try to assign it via "read_write_boolean", there is an error regarding "count does not match".

In comparison, if I adress it via the standard modbus plugin of home assistant like this it works:

```
switches:
        - name: "Wärmepumpe Betrieb"
          slave: 1
          address: 0x0320
          write_type: coil
          verify:
          unique_id: b8d74fc8-6d75-4983-ac32-8db5f6a3b136
```

To the other registers regarding flags. The map function works partially. Here is the extract of the PDF for one of them:

[image]

The value sent via modbus is the binary bit value, so from top to bottom 1, 2, 4, 8, 16, 32, 64, 128. That means if the message of bit 1 and bit 7 are active I get 130. So I would have to create a "map" for all possible combinations.
With the "flags" as I did in my earlier post. Then it works, except for my mentioned status of no message. Which sends "None" to Home Assistant and that Seems to "Invalid" for Home Assistant.

Other thing: If I adress each bit like this for example:

```
  hotwater_demand:
    name: "Heißwasser Nachfrage"
    address: 0x0003
    bits: 1
    device_class: heat
    control: binary_sensor
    icon: mdi:water-boiler
    entity_category: diagnostic
    entity_registry_enabled_default: True

  heating_demand:
    name: "Heizung Nachfrage"
    address: 0x0003
    bits: 1
    shift_bits: 1
    device_class: heat
    icon: mdi:heating-coil
    entity_category: diagnostic
    entity_registry_enabled_default: True

  antilegionella_on:
    name: "Antilegionellen Aktiv"
    address: 0x0003
    bits: 1
    shift_bits: 4
    device_class: running
    icon: mdi:virus-off
    entity_category: diagnostic
    entity_registry_enabled_default: True
```
That works, but as I said earlier, these are listed as "sensors" in Home assistant and not boolean. But I cannot assign control to binary_sensors for "read_write_word". That is blocked in your plugin.

timlaing, 2025-05-18:
@maniac-semal thanks for the clarification I will have a look at this for you.

maniac-semal, 2025-05-18:
Hello timlaing,

thanks for the help. On a sidenote regarding the switch which should be read as coil. I have been playing around with an External Tool. When using this, I can definately read and write adress 800 (or 320 in Hex) as a coil. Could it be that there is an issue in the read_write_boolean/read_only_boolean functions?
I noticed that in your predefined devices there is only maybe one which actually uses them.

maniac-semal, 2025-05-26:
Hello timlaing, 
Great job! Everything works as I hoped. I have some other things, but I will open new Issues for that to keep a propper follow-up.

Regards

closed 2025-05-26 by maniac-semal