timlaing/modbus_local_gateway #56: Feature to Invert State of Binary Sensors

maniac-semal, 2025-05-26:
Hello timlaing,

could it be possible to add an option to Invert a Binary Sensor? I have one Binary Sensor that issues its Status inverted:
```

  flow_switch:
    name: "Schalter Durchflussschalter"
    address: 0x0002
    bits: 1
    shift_bits: 4
    device_class: running
    control: binary_sensor
    entity_registry_visible_default: True
    entity_registry_enabled_default: True
```
In this case the flow switch is False-Active, so when the state is True, it means the Flow-Switch is off and vice versa. So it would be nice to have the possibility to invert the status directly here in config.

maniac-semal, 2025-06-08:
Hello timlaing,

could be, but I cannot find how to do it now. You nicely wrote it in the changelog that its a new Feature, but you seem to have forgotten to mention how in the readme.... :-)

timlaing, 2025-06-08:
@maniac-semal: In your example it would be

```yaml
  flow_switch:
    name: "Schalter Durchflussschalter"
    address: 0x0002
    bits: 1
    shift_bits: 4
    device_class: running
    control: binary_sensor
    entity_registry_visible_default: True
    entity_registry_enabled_default: True
    "on": False
    "off": True
```

maniac-semal, 2025-06-08:
Thank you! This works now.

closed 2025-06-08 by maniac-semal