timlaing/modbus_local_gateway #62: sum_scale: [65536, 1] with Little-endian results are wrong

Doorshifter, 2025-06-05:
### System Health details

## System Information

version | core-2025.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2001
Downloaded Repositories | 7

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.3
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 916.9 GB
disk_used | 27.1 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-m1
supervisor_api | ok
version_api | ok
[... 233 lines cut ...]
2025-06-05 14:32:06.388 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_current_outdoor_temperature_bt1_i_1_t): 1, count: 1
2025-06-05 14:32:06.388 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 1
2025-06-05 14:32:06.406 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_total_energy_i_2180_t): 2180, count: 2
2025-06-05 14:32:06.406 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 4 registers/coils from address 2180
2025-06-05 14:32:06.429 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_energy_lo): 2180, count: 1
2025-06-05 14:32:06.430 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 2180
2025-06-05 14:32:06.447 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_energy_hi): 2181, count: 1
2025-06-05 14:32:06.448 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 2181
2025-06-05 14:32:06.464 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Update completed AsyncModbusTcpClientGateway 192.168.2.201:502
2025-06-05 14:32:06.464 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_operating_mode_circulation_pump_heating_heat_pump_1_h_783_t is 0
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_current_outdoor_temperature_bt1_i_1_t is 19.8
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_total_energy_i_2180_t is 21509626068992.0
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_lo is 50081
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_hi is 50081
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Finished fetching Modbus Coordinator - 192.168.2.201:502:1 data in 0.093 seconds (success: True)
2025-06-05 14:32:06.465 DEBUG (MainThread) [custom_components.modbus_local_gateway.select] Updating device with nibe_operating_mode_circulation_pump_heating_heat_pump_1_h_783_t as Auto
2025-06-05 14:32:06.466 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value with nibe_current_outdoor_temperature_bt1_i_1_t as 19.8 - already set
2025-06-05 14:32:06.466 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Updating device with nibe_current_outdoor_temperature_bt1_i_1_t as 19.8
2025-06-05 14:32:06.466 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value with nibe_total_energy_i_2180_t as 21509626068992.0 - already set
2025-06-05 14:32:06.467 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Updating device with nibe_total_energy_i_2180_t as 21509626068992.0
2025-06-05 14:32:06.467 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value with nibe_energy_lo as 50081 - already set
2025-06-05 14:32:06.467 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Updating device with nibe_energy_lo as 50081
2025-06-05 14:32:06.467 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value with nibe_energy_hi as 50081 - already set
2025-06-05 14:32:06.467 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Updating device with nibe_energy_hi as 50081
```

timlaing, 2025-06-07:
This should be resolved in release v2025.6.0

Doorshifter, 2025-06-08:
it is getting better.
I have taken out the max_register_read: 1 
like this:

  nibe_total_energy_i_2180_t:
    name: Total energy t
    address: 2180
    size: 2
    multiplier: 0.1
    unit_of_measurement: "kWh"
    device_class: energy
    state_class: total_increasing
    precision: 1
    entity_category: diagnostic
    swap: word

after that I have seen a correct value for this sensor, but only briefly (so if you're too late you would think it doesn't work correctly 😄), just after the restart when the sensor becomes available. then when polling it jumps to 'nearby rounded values'. so from 18.211,6 kWh to 20.000,0 kWh. and then it stays like that for a very long time.

If I change the size: to 1. inch see the the same behavior. the first is correct (if you are fast) and the connectives are wrong and go to a 'rounded large number'.

dmatscheko, 2025-06-08:
I don't know about the strangely roundet numbers, but I can explain the behavior of sum_scale.

sum_scale already tells the integration, how many numbers it should scale and then summarize, so the "size: 2" is not necessary.

```
sum_scale: [1, 65536]
size: 2
```
would mean, take one number, consisting of 2 registers (because of size: 2), multiply it by 1. Then take the second number, consisting of the next 2 registers and multiply it by 65536. Then add the two results.

Look into Dimplex-SI-11TU.yaml for an example of how sum_scale can be used:
sum_scale: [1, 10000, 100000000]

In your case, `sum_scale: [1, 65536]` or `size: 2` should work, but not both combined.

timlaing, 2025-06-08:
> it is getting better. I have taken out the max_register_read: 1 like this:
>
> nibe_total_energy_i_2180_t: name: Total energy t address: 2180 size: 2 multiplier: 0.1 unit_of_measurement: "kWh" [...]

@Doorshifter - Are you able to provide me this debug logs for this, such that I can look at this for you.

Doorshifter, 2025-06-09:
Hi Tim,

I have taken this part of the some 2000 long list of registers that Nibe uses. It contains the register “Total energy”, in the middle of the list. I could if you want share the whole list, which is a standard Nibe export of all their registers if that is convenient for testing and understanding their data type usage. Don’t mind that min and max value are not as you would expect from certain registers. Sometimes these are correct in their export, and sometimes they are counter intuitive like min and max are 0. In practice int their world, this works and this seems to only be checked with MODBUS_HOLDING_REGISTER as type.

Title
Register type
Register
Division factor
Unit
Size of variable
Min value
Max value
Default value
Blocked compressors
MODBUS_INPUT_REGISTER
2174
1

u32
0
0
0
Current power
MODBUS_INPUT_REGISTER
2176
1
W
u32
0
0
0
Total average power (Solar PV)
MODBUS_INPUT_REGISTER
2178
100
kW
u32
0
0
0
Total energy
MODBUS_INPUT_REGISTER
2180
10
kWh
u32
0
0
0
Total oper. time ext. add. heat, heating
MODBUS_INPUT_REGISTER
2182
10
h
s32
0
9999999
0
Total HW run time additional heat
MODBUS_INPUT_REGISTER
2184
10
h
s32
0
9999999
0

I have commented out the native (working) modbus integration, to not interfere with the local modbus integration, although running them in parallel was not a problem and was used as a reference. The repairs on the automations in the debug log are there because my sync modbus automations are missing their counterparts.

[... 480 lines cut ...]
2025-06-09 07:00:53.169 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_total_energy_i_2180_t is 20000.0
2025-06-09 07:00:53.169 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_lo is 51059
2025-06-09 07:00:53.169 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_hi is 51059
2025-06-09 07:00:53.169 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Finished fetching Modbus Coordinator - 192.168.2.201:502:1 data in 0.057 seconds (success: True)
2025-06-09 07:00:53.170 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_total_energy_i_2180_t: 20000.0 – same as previous value
2025-06-09 07:00:53.170 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_energy_lo: 51059 – same as previous value
2025-06-09 07:00:53.170 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_energy_hi: 51059 – same as previous value
2025-06-09 07:01:23.112 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Updating data for Modbus Coordinator - 192.168.2.201:502:1 (AsyncModbusTcpClientGateway 192.168.2.201:502)
2025-06-09 07:01:23.112 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_total_energy_i_2180_t): 2180, count: 2
2025-06-09 07:01:23.112 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 2 registers/coils from address 2180
2025-06-09 07:01:23.122 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_energy_lo): 2180, count: 1
2025-06-09 07:01:23.123 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 2180
2025-06-09 07:01:23.146 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_energy_hi): 2181, count: 1
2025-06-09 07:01:23.146 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 2181
2025-06-09 07:01:23.156 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Update completed AsyncModbusTcpClientGateway 192.168.2.201:502
2025-06-09 07:01:23.157 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_total_energy_i_2180_t is 20000.0
2025-06-09 07:01:23.157 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_lo is 51059
2025-06-09 07:01:23.157 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Value for key nibe_energy_hi is 51059
2025-06-09 07:01:23.157 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Finished fetching Modbus Coordinator - 192.168.2.201:502:1 data in 0.045 seconds (success: True)
2025-06-09 07:01:23.157 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_total_energy_i_2180_t: 20000.0 – same as previous value
2025-06-09 07:01:23.158 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_energy_lo: 51059 – same as previous value
2025-06-09 07:01:23.158 DEBUG (MainThread) [custom_components.modbus_local_gateway.sensor] Ignoring device value for nibe_energy_hi: 51059 – same as previous value
2025-06-09 07:01:53.112 DEBUG (MainThread) [custom_components.modbus_local_gateway.coordinator] Updating data for Modbus Coordinator - 192.168.2.201:502:1 (AsyncModbusTcpClientGateway 192.168.2.201:502)
2025-06-09 07:01:53.112 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_total_energy_i_2180_t): 2180, count: 2
2025-06-09 07:01:53.112 DEBUG (MainThread) [custom_componen

Doorshifter, 2025-06-09:
The 32 bit registers don’t cause a problem in the standard home assistant modbus integration, I did some more investigating. If you are using the Home Assistant standard Modbus integration and find that **"swap: word"** (sometimes called "word swap" or "swap words") works correctly for both 16-bit and 32-bit registers with your NIBE SMO S40, this provides important clues about the device’s byte and word order:

- **"swap: word"** in Home Assistant’s Modbus integration means that for 32-bit values (which use two consecutive 16-bit registers), the order of the two registers is swapped before interpreting the data. This is often necessary when the SMO S40 transmits the lower word (lower 16 bits) first and the upper word (upper 16 bits) second, but the standard Modbus integration expects the opposite.
- **For 16-bit registers:** The "swap: word" setting should not affect interpretation, since each register is only 16 bits.
- **For 32-bit registers:** If "swap: word" makes the values correct, it means the SMO S40 is transmitting the two registers (words) in the "wrong" order for standard Modbus, but is using standard big-endian byte order within each 16-bit word.

**What this says about endianness:**
- **Within each 16-bit word (register):** The SMO S40 uses standard Modbus big-endian byte order (most significant byte first).
- **Between words (registers):** For 32-bit data, the SMO S40 sends the lower word first and the upper word second, which is **not** standard for most Modbus devices (which usually send the upper word first). This is why "swap: word" is needed to correct the order.
- **Summary:**  
  - **Byte order (within word):** Big-endian.
  - **Word order (between registers for 32-bit):** Lower word first, then upper word.

**In other words:**  
Your SMO S40 uses big-endian within each 16-bit register but reverses the order of the registers for 32-bit values. This is not standard Modbus, so you must use "swap: word" to correct it. This is not strictly about endianness (which refers to byte order), but about register order for multi-register values. However, in the context of Modbus and Home Assistant, this is commonly called "word swap."

timlaing, 2025-06-09:
This should be now resolved in the latest release

closed 2025-06-10 by Doorshifter