timlaing/modbus_local_gateway #63: Scan interval at register level iso device level

Doorshifter, 2025-06-05:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/timlaing/modbus_local_gateway/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

For heat pumps and other devices with a large number of registers—especially those used for both control and monitoring—it would be highly advantageous to allow the scan/update frequency to be defined on a per-register basis.

### Describe the solution you'd like

### Why per-register scan rates matter:
Some registers, such as countdown timers or fast-changing operational states, require high-frequency updates to ensure responsive control and monitoring.
Others, such as cumulative energy consumption or production values, change slowly and can be polled much less frequently without loss of information.
### Benefits:
Reduces unnecessary load on the Modbus server (the heat pump or device), which is primarily designed for process control, not for providing rapid-fire register updates to external clients.
Optimizes network and system resources by polling only what’s needed, when it’s needed.
Enhances reliability and responsiveness for critical controls, while minimizing the risk of overloading the device with excessive requests.
### In summary:
Per-register scan frequency would allow integrators to tailor polling rates to the nature of each register, improving both performance and device longevity.

### Describe alternatives you've considered

A potential workaround could be to define separate devices for each subsystem, assigning registers accordingly. However, this approach would result in a proliferation of devices and clutter the system with long lists of subsystems, making management and navigation less efficient and more cumbersome.

### Additional context

Home Assistant’s native Modbus integration already offers this capability, allowing users to set individual scan intervals and data handling options for each register or sensor.
see: https://www.home-assistant.io/integrations/modbus/

timlaing, 2025-06-05:
I will have a look at improving this integration with this feature. Thanks for your input.

timlaing, 2025-06-14:
This will be included in v2025.6.2