timlaing/modbus_local_gateway #73: Warning about unexpected keyword argument

maniac-semal, 2025-06-13:
### System Health details

## System Information

version | core-2025.6.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4987
Installed Version | 2.0.5
Stage | running
Available Repositories | 2086
Downloaded Repositories | 90

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.5
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 219.4 GB
disk_used | 43.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | AppDaemon (0.17.4), BirdNET-pi (2025.04.08), EMQX (0.7.5), Git pull (8.0.1), MQTT Explorer (browser-1.0.1), Music Assistant Server (2.5.4), NSPanel Manager (1.0.4), Piper (1.5.4), SQLite Web (4.3.1), Terminal & SSH (9.17.0), Zigbee2MQTT (2.4.0-1), chrony (5.1.0), vosk (1.6.1), microwakeword (1.0.0), Firefly iii (6.2.17), Firefly iii FinTS Importer (1.3.0-4), ESPHome Device Builder (2025.5.2), Silicon Labs Flasher (0.3.2), Matter Server (8.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 50
views | 17
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), eno1 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth45992ab (disabled), vethe3215b7 (disabled), veth2ab82c2 (disabled), vethd0761e7 (disabled), veth39e9341 (disabled), veth909d15c (disabled), veth5ec1b4f (disabled), veth5d4d8a4 (disabled), vetha27c7fc (disabled), veth797a77c (disabled), veth8a9aef3 (disabled), vethfeaf824 (disabled), vethb08e3c4 (disabled), veth3b9b530 (disabled), vethab66416 (disabled), vethc5b5f04 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), eno1 (10.18.2.1/16), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth45992ab (), vethe3215b7 (), veth2ab82c2 (), vethd0761e7 (), veth39e9341 (), veth909d15c (), veth5ec1b4f (), veth5d4d8a4 (), vetha27c7fc (), veth797a77c (), veth8a9aef3 (), vethfeaf824 (), vethb08e3c4 (), veth3b9b530 (), vethab66416 (), vethc5b5f04 ()
ipv6_addresses | lo (::1/128), eno1 (fe80::cdb9:3d17:d901:8905/64), hassio (fe80::4864:feff:fe9a:c97c/64), docker0 (fe80::942e:a6ff:fe4c:a776/64), veth45992ab (fe80::8fa:1ff:fe75:845b/64), vethe3215b7 (fe80::c073:b3ff:fe34:a328/64), veth2ab82c2 (fe80::f0ab:e1ff:fe9c:47b0/64), vethd0761e7 (fe80::c4e3:b9ff:fe8c:76c8/64), veth39e9341 (fe80::470:25ff:fe03:6e7e/64), veth909d15c (fe80::3cb4:95ff:feb6:2562/64), veth5ec1b4f (fe80::3cb1:feff:fe52:a90c/64), veth5d4d8a4 (fe80::20fe:10ff:fe12:b5b/64), vetha27c7fc (fe80::34bf:6aff:fe38:1c93/64), veth797a77c (fe80::a419:b2ff:fea7:4ed5/64), veth8a9aef3 (fe80::60fa:8cff:fe28:1d6e/64), vethfeaf824 (fe80::4c9c:b1ff:fe71:180a/64), vethb08e3c4 (fe80::70c3:efff:fefa:f1/64), veth3b9b530 (fe80::84f2:59ff:fed9:4a79/64), vethab66416 (fe80::9421:b0ff:feaf:2e53/64), vethc5b5f04 (fe80::ecf8:c5ff:fe75:aba2/64)
announce_addresses | 10.18.2.1, fe80::cdb9:3d17:d901:8905

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2. Juni 2025 um 20:10
-- | --
current_recorder_run | 14. Juni 2025 um 00:33
estimated_db_size | 1361.88 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/timlaing/modbus_local_gateway/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I am getting a Warnung from Home assistant in my logs in regards to a "unexpected keyword argument"

### Reproduction steps

After every restart of Home Assistant

### Debug logs

```text
Logger: custom_components.modbus_local_gateway.entity_management.modbus_device_info
Quelle: custom_components/modbus_local_gateway/entity_management/modbus_device_info.py:383
Integration: Modbus Local Gateway (Dokumentation, Probleme)
Erstmals aufgetreten: 00:33:39 (12 Vorkommnisse)
Zuletzt protokolliert: 00:33:39

Failed to create description instance for <class 'custom_components.modbus_local_gateway.entity_management.base.ModbusSelectEntityDescription'>: ModbusSelectEntityDescription.__init__() got an unexpected keyword argument 'native_unit_of_measurement'. Did you mean 'unit_of_measurement'?
```

Doorshifter, 2025-06-14:
### Problem
The code is passing **native_unit_of_measurement** as a keyword argument to ModbusSelectEntityDescription.__init__(), but this is not a valid argument for that class. It does this 12x
### Suggestion
The error message you received guesses (you did not provide the config yaml) that you may have meant **unit_of_measurement** instead.

### Why This Happens
### Sensor vs. Select Entity example
**Sensor entities** in Home Assistant use native_unit_of_measurement for **internal** value handling and unit_of_measurement for **display**.  I agree this is a bit of a pitfall in the HA documentation.
**Select entities** (like dropdowns) do not have units of measurement, so they **should not be given** either native_unit_of_measurement or unit_of_measurement as arguments.

### To solve this
Either **remove** the lines with the keyword native_unit_of_measurement from your device configuration if they are in the **wrong place** (see above),save the file, followed by a HA restart or,.....
Do a '**find and replace**' with native_unit_of_measurement **replacing** it with unit_of_measurement if your units need definition but you misinterpreted the HA doc, save the file, followed by a HA restart.

maniac-semal, 2025-06-14:
Hello Doorshifter,
If you read the error message I printed above, you will notice that it is referring to a file of the addon itself.... Therme is in fact the line with "native" there... 

`custom_components/modbus_local_gateway/entity_management/modbus_device_info.py:383`
I don't have the "native" in my config.... First Place I checked.

commit 5a5b3e61 (2025-06-14): Rename 'native_unit_of_measurement' to 'unit_of_measurement' in ModbusDeviceInfo for consistency (#74)

Resolves #73
closed 2025-06-14 by timlaing