timlaing/modbus_local_gateway #79: Invalid response received from slave 1, address: 1588, .... (count does not match)

Doorshifter, 2025-06-14:
### System Health details

## System Information

version | core-2025.6.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 2022
Downloaded Repositories | 7

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.5
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 916.9 GB
disk_used | 28.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | odroid-m1
supervisor_api | ok
version_api | ok
[... 276 lines cut ...]
2025-06-14 16:45:45.872 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_556): 556, count: 1
2025-06-14 16:45:45.873 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 556
2025-06-14 16:45:45.881 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_557): 557, count: 1
2025-06-14 16:45:45.881 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 557
2025-06-14 16:45:45.894 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_579): 579, count: 1
2025-06-14 16:45:45.895 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 579
2025-06-14 16:45:45.904 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_581): 581, count: 1
2025-06-14 16:45:45.904 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 581
2025-06-14 16:45:45.914 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_582): 582, count: 1
2025-06-14 16:45:45.914 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 582
2025-06-14 16:45:45.925 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_590): 590, count: 1
2025-06-14 16:45:45.925 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 590
2025-06-14 16:45:45.934 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_591): 591, count: 1
2025-06-14 16:45:45.935 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 591
2025-06-14 16:45:45.945 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_592): 592, count: 1
2025-06-14 16:45:45.945 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 592
2025-06-14 16:45:45.956 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_593): 593, count: 1
2025-06-14 16:45:45.956 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 593
2025-06-14 16:45:45.966 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_683): 683, count: 1
2025-06-14 16:45:45.966 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 683
2025-06-14 16:45:45.980 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_684): 684, count: 1
2025-06-14 16:45:45.980 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 684
2025-06-14 16:45:45.995 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Reading slave: 1, register/coil (nibe_input_register_685): 685, count: 1
2025-06-14 16:45:45.996 DEBUG (MainThread) [custom_components.modbus_local_gateway.tcp_client] Trying to read 1 registers/coils from address 685
```

Doorshifter, 2025-06-14:
While scrolling through today’s list of sensor values I noticed that sensor.nibe_compressor_size_eb101 occasionally changes from 12 to 17. That would mean that my heat pump suddenly increased from its 12 kW to 17 kW. That would be magic 😊. But it has probably something to do with the interpretation of the values coming back from the system. The current read amount of registers is the default 8.

Doorshifter, 2025-06-14:
Reopen

timlaing, 2025-06-14:
@Doorshifter - I'll take a look at this later this week

Doorshifter, 2025-06-21:
I tried to continue with this and it seems more complex than I thought it would be 😄

timlaing, 2025-06-21:
Thank you for your input. When I get a few minutes this weekend I’ll see what I can do for you.

Doorshifter, 2025-06-21:
😊Gr. M

closed 2026-10-04 by timlaing